imuncle/imuncle.github.io #1: issue测试

imuncle, 2019-02-05:
第一条issue

imuncle, 2019-02-05:
针对issue的评论

imuncle, 2019-02-05:
@imuncle 提醒功能测试

imuncle, 2019-02-07:
comments api test

imuncle, 2019-02-08:
test again

imuncle, 2019-02-09:
markdown test

header|header
:--:|:--:
content|content

imuncle, 2019-02-10:
代码高亮测试

```javascript
function test() {
    alert('代码高亮测试');
}
```

imuncle, 2019-02-10:
JavaScript脚本语言不解析测试

alert('这条语句不会执行');

imuncle, 2019-02-10:
Tex公式测试

$$a^2=b^2+c^2$$

imuncle, 2019-02-10:
图片测试

[image]

imuncle, 2019-02-10:
视频测试

<iframe src="//player.bilibili.com/player.html?aid=23318839&cid=38830500&page=1" scrolling="no" border="0" frameborder="no" framespacing="0" allowfullscreen="true"> </iframe>

imuncle, 2019-02-10:
随便凑个数，测试分页功能

imuncle, 2019-02-10:
随便凑个数，测试分页功能

imuncle, 2019-02-13:
最后一次测试

imuncle, 2019-02-13:
LaTeX公式测试

![](https://latex.codecogs.com/gif.latex?$$a^2=b^2&plus;c^2$$)

imuncle, 2019-02-23:
To list 测试

- [X] Pick up rental car
- [ ] Finish essay
- [ ] Finish essay
- [ ] Finish essay
- [ ] Finish essay

imuncle, 2019-06-01:
儿童节-代码架构更改-评论测试

imuncle, 2019-06-01:
评论测试

imuncle, 2019-06-23:
公式测试

$$E = mc^2$$

goofegg, 2022-05-18:
这个页面好炫，而且点赞可以计数的呢。

goofegg, 2022-05-18:
这里的视频测试不正常哦？点赞，选图标那个怎么实现的？加到页面里很有趣