imuncle/imuncle.github.io #22: 网站改造总结

imuncle, 2019-02-13:
前几天春节期间花了点时间把个人网站改造了一下，在这里做个总结Mark一下。

改造的念头来源于一次无意间翻到的一个在issue页面建立的博客（相当于只有后台的博客），我就有了要将网站与issue挂钩的想法。

之前我的网站是基于[Editor.md](https://pandao.github.io/editor.md/examples/)的，以在本地写好相应的.md文件后push到GitHub上的方式来发表文章，也不是太麻烦，唯有一个遗憾，就是没有评论功能。去年的时候我接触过[Gitment](https://github.com/imsun/gitment)，但当时还不太了解GitHub，更别说issue了，而且Gitment需要手动为每个文章初始化一次，显得比较麻烦。现在作者网站的安全证书过期了，导致Gitment访问不是很稳定。

于是我决定自己来试着把GitHub的issues和comments都集成到我的网站里。

首先是注册一个GitHub的OAuth App，点[这里](https://github.com/settings/applications/new)可以注册。注册后可以拿到对应的`client_id`和`client_secret`。

最开始我一直想通过纯JavaScript实现此功能，但总是遇到各种各样的问题，最后遇到了一个实在无解的问题，详情可以点[这里](?id=21)，然后我发邮件询问GitHub，结果直接来了个官方劝退：

> Exchanging a code for a token is not possible from a browser via JavaScript -- it's not something that our OAuth flow supports and you'll run into several problems stopping you if you try doing that. Instead, you'll need to have a server-side component which makes that request for exchanging the code for a token.

看来只能再来个服务端的程序了，怪不得Gitment老是要访问作者的那个网址。

我使用了最熟悉的PHP，代码很简单，如下：

```php
<?php
$code = $_GET['code'];
$redirect_url = $_GET['state'];

$client_id = 'my client_id';
$client_secret = 'my client_secret';

$url = "https://github.com/login/oauth/access_token?client_id=".$client_id."&client_secret=".$client_secret."&code=".$code;
 
$html= file_get_contents($url);

$data = explode("&", $html);
$token = explode("=", $data[0]);
?>
<script>
window.location.href = '<?php echo $redirect_url; ?>&access_token='+'<?php echo $token[1]; ?>';
</script>
```

在获取到access_token后，我拿着这个token进一步获取到了登录用户的用户名和头像。

我直接使用了Gitment的css样式，具体样式可以直接拉到文章底部查看。

这里还有一个坑。其实也不算是坑，只是我不太了解OAuth，相关注意事项全都写在文档里面的。

我开始申请到access_token后，尝试添加评论，但是老是提示我`Must have admin rights to Repository.`，header提示`403 Forbidden`，这是没有访问repository的意思，后来我才知道，原来申请code的时候要在get参数里面加上'scpoe=public_repo`，才能授予用户修改的权限。

最后我又完善了一些分页功能，首页以列表形式展示issue的时候去除HTML标签。

然后就可以体验全新的网站啦。

关于这些功能的实现具体可以查看我的源代码[issue-comment.js](https://github.com/imuncle/imuncle.github.io/blob/master/js/issue-comment.js)。

hitszqmaker, 2019-02-14:
@imuncle nice！学习了！:smile:

顺便测试emoji 💯

vaeann, 2019-12-10:
server_link 必须需要自己的吗 还是使用默认提供的也可以

imuncle, 2019-12-10:
@vaeann 可以使用默认提供的，但不保证哪天更换服务器，因为我的学生优惠快到了。。

vaeann, 2019-12-10:
我是个没有基础的小白，操作完只出现主页没有文章之类的 麻烦您有空帮我看一下排除下问题，起初我以为是服务器link的问题 看来不是 地址是 vaeann.githun.io

vaeann, 2019-12-10:
手机打字打错了 vaeann.github.io

imuncle, 2019-12-10:
@vaeann 加我QQ吧，点最下面的QQ图标就能跟我聊天