imuncle/imuncle.github.io #39: MPU9250姿态解析

imuncle, 2019-03-18:
前段时间尝试使用陀螺仪作为机器人云台的位置环反馈值，在底盘跟随的大前提下，这样做是更符合第一视角的操作的。

我去年比赛的时候买了几个维特智能的型号为WT901的九轴，当时时间紧没有用上，结果不用不知道，一用吓一跳，陀螺仪延迟高的吓人，使用J-Scope看到的图像如下：
[image]

从图中可以看出延迟大约有200ms，这种延时已经不再具备方位参考值了。

我在仓库里东翻西找，找到了两个三十多块的MPU9250，相比于我买的一百来块的陀螺仪来说便宜了很多，于是我开始尝试手动解析数据。

# 读取原始数据
MPU9250内部其实是由6轴陀螺仪**MPU6500**和磁力计**AK8963**组成，可以测量三轴加速度，三轴角速度和三轴磁场强度，此外还可以测量温度，但这里我没有用到温度的数据。

数据解析的第一步就是读取原始数据。MPU6500支持IIC通信和SPI通信，AK8963只支持IIC通信，但可以通过MPU9250为中介使用SPI通信，只是比较麻烦，所以这里我选择了IIC通信。

## IIC地址
MPU9250的IIC地址取决于AD0口的电平高低，如果AD0接GND，则地址为0x68，如果接VCC，则地址为0x69,。

这里说一下一个坑：
> 在读取时，注意需要人工将地址左移1位（I2C读写为左对齐，第8位要存读写标志位）

最开始我不知道这个，一直把地址写成0x68，结果死活读不到数据，后来才发现要写成0xD0。

MPU9250的其他地址可以查看它的寄存器数据手册，我这里列出了我用到的一部分寄存器：

```c
#define MPU9250_ADDRESS 0xD0   //AD0接GND时地址为0x68，接VCC时地址为0x69
#define MPU_PWR_MGMT1_REG		0X6B	//电源管理寄存器1
#define MPU_GYRO_CFG_REG		0X1B	//陀螺仪配置寄存器
#define MPU_ACCEL_CFG_REG		0X1C	//加速度计配置寄存器
#define MPU_SAMPLE_RATE_REG		0X19	//陀螺仪采样频率分频器
#define MPU_INT_EN_REG			0X38	//中断使能寄存器
#define MPU_USER_CTRL_REG		0X6A	//用户控制寄存器
#define MPU_FIFO_EN_REG			0X23	//FIFO使能寄存器
#define MPU_INTBP_CFG_REG		0X37	//中断/旁路设置寄存器
#define MPU_DEVICE_ID_REG		0X75	//器件ID寄存器
#define MPU_PWR_MGMT2_REG		0X6C	//电源管理寄存器2
#define MPU_CFG_REG				0X1A	//配置寄存器 低通滤波器配置寄存器

#define MPU_TEMP_OUTH_REG		0X41	//温度值高8位寄存器
#define MPU_TEMP_OUTL_REG		0X42	//温度值低8位寄存器

#define MPU_ACCEL_XOUTH_REG		0X3B	//加速度值，X轴高8位寄存器
#define MPU_ACCEL_XOUTL_REG		0X3C	//加速度值，X轴低8位寄存器
#define MPU_ACCEL_YOUTH_REG		0X3D	//加速度值，Y轴高8位寄存器
#define MPU_ACCEL_YOUTL_REG		0X3E	//加速度值，Y轴低8位寄存器
#define MPU_ACCEL_ZOUTH_REG		0X3F	//加速度值，Z轴高8位寄存器
#define MPU_ACCEL_ZOUTL_REG		0X40	//加速度值，Z轴低8位寄存器

#define MPU_GYRO_XOUTH_REG		0X43	//陀螺仪值，X轴高8位寄存器
#define MPU_GYRO_XOUTL_REG		0X44	//陀螺仪值，X轴低8位寄存器
#define MPU_GYRO_YOUTH_REG		0X45	//陀螺仪值，Y轴高8位寄存器
#define MPU_GYRO_YOUTL_REG		0X46	//陀螺仪值，Y轴低8位寄存器
#define MPU_GYRO_ZOUTH_REG		0X47	//陀螺仪值，Z轴高8位寄存器
#define MPU_GYRO_ZOUTL_REG		0X48	//陀螺仪值，Z轴低8位寄存器

#define AK8963_ADDRESS 0x18		//磁力计地址0x0C
#define AK8963_CNTL1		0x0A  //磁力计读取寄存器
#define AK8963_HXL			0x03	//磁力计X轴低8位寄存器
#define AK8963_HXH			0x04	//磁力计X轴高8位寄存器
#define AK8963_HYL			0x05	//磁力计Y轴低8位寄存器
#define AK8963_HYH			0x06	//磁力计Y轴高8位寄存器
#define AK8963_HZL			0x07	//磁力计Z轴低8位寄存器
#define AK8963_HZH			0x08	//磁力计Z轴高8位寄存器
```

## 配置MPU9250
读取数据之前要对MPU9250进行一系列的配置，比如设置采样频率，设置量程等等，具体的如下：

```c
void MPU9250_Init(void)
[... 229 lines cut ...]
	q1 += ( q0 * gyro.x + q2 * gyro.z - q3 * gyro.y) * halfT;
	q2 += ( q0 * gyro.y - q1 * gyro.z + q3 * gyro.x) * halfT;
	q3 += ( q0 * gyro.z + q1 * gyro.y - q2 * gyro.x) * halfT;
	
	/* 对四元数进行归一化处理 */
	normalise = sqrt(q0 * q0 + q1 * q1 + q2 * q2 + q3 * q3);
	q0 /= normalise;
	q1 /= normalise;
	q2 /= normalise;
	q3 /= normalise;
	
	/* 由四元数求解欧拉角 */
	mpu9250.attitude.x = -asinf(-2*q1*q3 + 2*q0*q2) * RAD2DEG;	//pitch
	mpu9250.attitude.y = atan2f(2*q2*q3 + 2*q0*q1, -2*q1*q1 - 2*q2*q2 + 1) * RAD2DEG;	//roll
	mpu9250.attitude.z = atan2f(2*q1*q2 + 2*q0*q3, -2*q2*q2 - 2*q3*q3 + 1) * RAD2DEG;	//yaw
	
	yaw = mpu9250.attitude.z*100;		//用于J-Scope读取
	pitch = mpu9250.attitude.x*100;
	roll = mpu9250.attitude.y*100;
}
```

# 参考
1. [【教程】四旋翼飞行器姿态解算算法入门学习-Rick Grimes](https://www.bilibili.com/video/av13035245?from=search&seid=6244828450901036366)
2. [[UVA]Pixhawk之姿态解算篇（2）](http://www.07net01.com/2016/03/1398642.html)

LUCKand, 2019-03-18:
太强啦！

imuncle, 2019-03-18:
附上最后的效果：
[image]
红线是电机码盘，绿线是陀螺仪数据（yaw轴），几乎完全重合，并且无漂移

pinocchiolhw, 2019-07-26:
谢老板🐮b

zcf0619, 2021-01-15:
博主你好，我想参考一下您关于磁力计数据读取函数的代码可以吗，提前谢谢啦。

zcf0619, 2021-01-20:
前辈你好，如果可以的话可以分享一下您完整的工程吗。

imuncle, 2021-01-25:
@zcf0619 在这里： https://github.com/imuncle/Embedded_Peripheral_Libs

zcf0619, 2021-01-25:
你好前辈，非常感谢！

zcf0619, 2021-01-28:
前辈你好，
冒昧致函。想向您请教在AHRS解算中，Kp和Ki参数的大致范围应该如何确定呢。
望不吝赐教。

赵存飞

北京航空航天大学宇航学院  图像处理中心
通讯地址：北京市昌平区沙河高教园南三街9号
邮政编码：102206
在2021年1月25日 11:42，赵存飞<17732726639@163.com> 写道：

你好前辈，非常感谢！