imuncle/imuncle.github.io #68: 通过链接形式唤起QQ对话

imuncle, 2019-06-26:
通过下载百度网盘资料时浏览器会提示打开百度网盘程序，淘宝咨询客服时会提示打开阿里旺旺程序，显然像这种很有名的软件可能会留有官方的渠道，可以通过一个链接唤起系统的其他程序。

于是突然想实现在网页上点击链接就能直接开启QQ聊天界面的功能。在各种寻找之后，终于在[这篇文章](https://idealclover.top/archives/545/)找到了链接。

使用的是QQ推广的官方渠道，可以使得无论网页端还是手机端，无论QQ还是TIM，都可以自动调用相关应用并联系到我。

```javascript
http://wpa.qq.com/msgrd?v=3&uin=QQ号&site=qq&menu=yes
```

```javascript
tencent://message/?uin=QQ号
```

上面两个链接都可以使用。比如
 
[点我联系作者](http://wpa.qq.com/msgrd?v=3&uin=1378006772&site=qq&menu=yes)

Crack-Yi, 2019-06-26:
Tim电脑端好像不行的哦！

imuncle, 2019-06-27:
@Crack-Yi  试试这个 [image]