imuncle/imuncle.github.io #73: 四足机器人制作 （五） 代码实现

imuncle, 2019-07-21:
距离上一篇文章已经有一个半月了，越咕越得劲（其实是因为小学期整天上实验课啦），前天刚结束小学期的课程，觉得是时候给这个小项目结个尾了。

我花了四个小时设计机械结构，十二个小时接线，三个小时写代码，九个小时改错调试，最后的作品长这样：
[image]

（~~这TM也太丑了吧~~）

因为能力有限，机械结构的刚度也不够，前面文章所设想的部分功能未能实现，深表遗憾。

骚话不多说，直接进正题。

# 腿部运动
我使用的是飞特的[SCS0009舵机](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.66582e8dmATkeS&id=539949416714&_u=l2osqi7a9c5d)，体积很小，扭矩足够，串行总线，简直不要太爽！

因为舵机个体以及舵机安装方式的问题，同一个位置信号发过去，不同的舵机反应不一样，所以最开始我们需要校准。

我记录的是四条腿都伸直的时候的舵机位置反馈值，存储在一个数组里面：
```c
int init_position[12] = {550, 641, 796,
			 525, 491, 79,
			 600, 864, 766,
			 274, 342, 207};
```

当然小狗是不能保持伸直的，所以我在初始化的时候给了一个足尖初始坐标(0,0,86.6)，坐标系的建立见后面的图。

从前面的文章（[四足机器人制作（一） 腿部运动算法](https://imuncle.github.io/content.html?id=61)）可以看出，我的腿部控制逻辑为：先确定足尖的坐标，然后根据几何关系求解出腿部三个舵机的角度，最后转换为对应的舵机位置信号。先上代码：
```c
void LegCalc(struct Leg_t* leg)
{
	leg->x =  leg->delta_x;
	leg->y =  leg->delta_y;
	leg->z =  86.6f + leg->delta_z;
	float L;
	L = sqrt(leg->x * leg->x + leg->y * leg->y + leg->z * leg->z);
	leg->angle_1 = asin(leg->y/L);
	leg->angle_2 = acos(L/100) - asin(leg->x/L);
	leg->angle_3 = acos(1 - L*L/5000);
	leg->angle_1 *= RAD_TO_DEGREE;
	leg->angle_2 *= RAD_TO_DEGREE;
	leg->angle_3 *= RAD_TO_DEGREE;
	leg->angle_3 = 180 - leg->angle_3;
	leg->position[0] = leg->bias_position[0] + leg->coefficient_1 * leg->angle_1 * ANGLE_TO_ENCODER;
	leg->position[1] = leg->bias_position[1] + leg->coefficient_2 * leg->angle_2 * ANGLE_TO_ENCODER;
	leg->position[2] = leg->bias_position[2] + leg->coefficient_2 * leg->angle_3 * ANGLE_TO_ENCODER;
}
```
这里写的有点不规范，我为了图方便直接把我的狗腿长度带入计算公式化简了。我的小狗大腿和小腿长度都是50mm。

模型如下（注意坐标系的原点和坐标轴正方向）：
[image]

其中AF和FD长度都是50mm，可以通过解三角形得到，α（对应代码中angle_2），β（angle_3），γ（angle_1）的计算公式如下：

$$\gamma = arc\sin(\frac{AD}{CD})$$
$$\alpha = arc\cos(\frac{AD^2+AF^2-DF^2}{2AD·AF}) - arc\sin(\frac{DE}{AD})$$
$$\beta = 180 - arc\cos(\frac{AF^2+FD^2-AD^2}{2AF·FD})$$

把坐标和腿长带进去，就得到上面的代码。

上面的代码中，`delta_x`之类的变量记录的是机器人各种变化（比如走路）对单条腿产生的坐标变化值，将它叠加在腿的初始位置上就得到了这个时刻腿的目标位置。根据这个位置计算完角度后，将角度换算为舵机的位置信息（最后三行），这里有变量`coefficient_1`和`coefficient_2`，这是因为在机械结构上，不同的舵机安装方式不同，对于“**角度增大**”这一指令，它既可能逆时针旋转，也可能顺时针旋转，所以在初始化的时候为每条腿设置了对应的系数（1或-1），拿右前腿为例，它的初始化代码如下：
```c
void BodyInit(void)
{
	FR_Leg.IDs[0] = 1;
	FR_Leg.IDs[1] = 2;
	FR_Leg.IDs[2] = 3;
	FR_Leg.bias_position[0] = init_position[0];
	FR_Leg.bias_position[1] = init_position[1];
	FR_Leg.bias_position[2] = init_position[2];
[... 258 lines cut ...]
	}
	if(leg->state != leg->last_state) leg->count = 0;
	leg->count++;
	if(leg->state == Up)
	{
		leg->delta_x = -StepLength+leg->count*2*StepLength/cycle;
	}
	else if(leg->state == Down)
	{
		leg->delta_x = StepLength-leg->count*2*StepLength/(3*cycle);
	}
}
```
从上面的代码中可以看出，在腿抬起阶段（Up），腿从`-StepLength`匀速移动到`StepLength`，在放下阶段（Down），又从`StepLength`匀速移动到`-StepLength`，实现了前进移动。

横移和转弯也是类似，不做赘述。

# 云台稳定
在我之前的设想中，这一部分是整个机器人最复杂的部分，也是整个机器人最亮眼的地方，然而经过一番缜密的推算后发现，我之前提出的想法（惯导实现室内定位部分）无法实现，原因是传感器数据不够，本来想着在狗头上加一个光流传感器算了，但是千不该万不该我当初买舵机少买了一个，导致光流没地方装（现在就是后悔，非常后悔.jpg）

但最基本的机器狗原地稳定狗头还是能做的嘛。

我们需要解决两个轴：yaw和pitch，roll轴由于机械方面的限制而无法实现。我们一个一个来。
## yaw
写累了休息一会……

imuncle, 2019-07-21:
源代码地址：https://github.com/imuncle/RobotDog

imuncle, 2019-07-26:
看了机器人学后颇有感触，有空把矩阵用在四足上

zhuangfengzi, 2019-07-27:
很棒！
很期待！

FENYUN323, 2019-07-30:
太强了

garrisonz, 2026-03-06:
> 源代码地址：https://github.com/imuncle/RobotDog
这个网站的数据加载不出来，一直是 “内容加载中...”