imuncle/imuncle.github.io #81: 嵌入式视觉OpenOV研发（一） 初版PCB设计

imuncle, 2019-10-15:
好几周没写文章了，主要是因为保研了嘛，又被学弟叫回战队去了，前段时间在忙着搞培训，制定规章制度之类的，当然最主要的还是因为我想搞一个大的。

今年的RoboMaster多了很多元素，据说今天出规则，不过现在我写文章的这个时刻还没出，估计再过几个小时就出了吧。今年主要多了导弹和雷达站元素，其中导弹需要嵌入式视觉的参与，于是我就参与了嵌入式视觉的研发。而我最熟悉的还是MCU的开发，所以希望能直接在MCU上运行视觉代码。

# openmv
目前市面上最火爆的，运行在MCU上的视觉识别产品是[openmv](https://openmv.io/)，它是完全开源的，github地址在[这里](https://github.com/openmv/openmv)，目前国内的官方代理商是[星瞳科技](https://singtown.com/openmv/)。

星瞳科技做的比较好的一点是把官方的openmv英文文档翻译成了中文，不过有些地方翻译的还不是太好。英文文档点[这里](http://docs.openmv.io/)，翻译的中文文档点[这里](https://docs.singtown.com/micropython/zh/latest/openmvcam/index.html)。

openmv内嵌了[micropython](https://github.com/micropython/micropython)，实现了在单片机上运行python，支持最新的python 3.X，还是非常不错的。回想起来前几年在MakerFair上看到的可以用python编程的单片机，估计就是内嵌了micropython，亏我当时还以为是那些公司自己写的中间层，没想到是直接用的开源的东西。

正是因为有了micropython，openmv留给用户的接口是python的接口，且基于Qt开发了一个挺舒服的IDE，用户直接在里面用python调用API就行了，上手非常快，效果看得见。目前主要是各种电子设计大赛中会见到它的身影。

那么既然已经有了openmv为什么还要自己另外做嵌入式视觉呢？原因主要有以下几点：
* openmv留出来的接口都是固定的，虽说已经很多了，但是可能在一些很特殊的环境下无法很好完成工作
* openmv内部是封装好的，优化是没法优化的
* 用户代码用python写，写好之后是以字符串的形式传给单片机的，单片机还得挨个解析，然后调用相应的函数，资源再怎么也耗费了一些，感觉不爽
* 要不是openmv用的 F7和H7系列，性能绝对不行
* 最重要的一点，直接用openmv多无趣啊，RM比赛本就是挑战自我的，正好趁这次机会学点新东西

综上，我选择了自己研发。

# OpenOV
先给自己的东西取个名字。其他两搞视觉的一个是`opencv`，一个是`openmv`，而我的打算用OmniVision公司的OV系列摄像头（其实openmv用的也是这个），所以我的就叫做`OpenOV`吧。

先列一下我所计划有的资源：
* 摄像头×1
* LCD×1
* TF卡槽×1
* LED指示灯×2
* Jlink下载器集成
* USB口×2

预计将要实现以下功能：
* 图像采集和显示
* 基本的图像处理，包括图像二值化、各种滤波、快速傅里叶变换、直方图转换、霍夫变换提取特征、canny边缘识别等，要是还有精力可以考虑实现一下级联分类器实现人脸识别之类的
* TF卡读写，可存储图像或识别用的匹配模板等
* 读卡器功能，通过USB口模拟读卡器，快速方便操作SD内的文件

就先规划这么多吧。

## PCB设计
首先进行PCB设计。我使用的是AD，原理图很简单：
[image]

开发板的尺寸也很迷你，大约是3cm×5cm。最后焊接组装后的成品如下：
[image]

## 单片机外设选型
目前只是方案探索阶段，所以没有直接上高性能的MCU，而是直接延用了机器人上的STM32F405RGT6，不得不说资源太少了，不仅没有摄像头专用的DCMI，连驱动LCD的FSMC也没有，真的太惨了。

鉴于以上几点，最后我选择了一款SPI通信的0.96寸的LCD，摄像头选择了正点原子的带有FIFO的OV7725，实际效果还行，但帧率肯定比不过用DCMI驱动的摄像头。

考虑到F4的性能问题，同时也是为了适应小LCD，我最后把图像的输出尺寸降到了160×60，真的是非常小的图片了，希望F4能胜任。

# 未完待续
这只是OpenOV的第一步，基本的电路板测试无误后，接下来就是写代码了。

LUCKand, 2019-10-15:
马上来打call！

FENYUN323, 2019-10-15:
tql

FENYUN323, 2019-10-15:
tql

ws2019qh, 2019-11-01:
>T
>>q
>>>l