imuncle/imuncle.github.io #83: 自制CMSIS-DAP下载器

imuncle, 2019-10-18:
上周末硬件组的人说想搞个无线下载器，就用ARM开源的CMSIS-DAP协议，结果就这一句话我自闭了四五天，终于在一篇非常棒的文章里找到了方法，因为写的太好了，所以这里我直接把原文翻译成中文，英文原文见：[usb_express:cmsis-dap [Geniekits]](http://wiki.geniekits.com/doku.php?id=usb_express:cmsis-dap)

# 在STM32F103C8上实现CMSIS-DAP
* 本工程的相关代码已经上传至[Github](https://github.com/GenieKits/CMSIS-DAP_on_STM32_with_EmBitz)，你也可以从我的[下载页面](http://wiki.geniekits.com/downloads)获取到源文件。

---
如果你想在主流的MCU上进行嵌入式开发，一般来说你需要一个JTAG来连接单片机和电脑。对于ARM的MCU来说，[J-Link](https://www.segger.com/products/debug-probes/j-link/)是一个非常有名的设备，它可以提供JTAG协议将单片机个电脑连接起来。CMSIS-DAP是一个由ARM公司发布的一款类似的设备。DAP和J-Link最主要的区别在于DAP使开源的，而J-Link是一个付费产品。另外，CMSIS-DAP是基于USB HID的，所以你不需要额外安装一个固件驱动就能使用。

CMSIS-DAP 2.0.0的源码在[CMSIS V5](https://github.com/ARM-software/CMSIS_5)的GitHub仓库里。在源码里还有两个基于[LPC-Link-II](https://www.nxp.com/design/microcontrollers-developer-resources/lpc-microcontroller-utilities/lpc-link2:OM13054)的示例。这两个示例使用的是[Keil MDK](http://www.keil.com/)的[RTE组件和USB库](http://www.keil.com/support/man/docs/uv4/uv4_ca_swcomponents.htm)。在源码中还有一个叫做`USB_CM3.lib`的库文件。正因为这些，官方提供的这个示例很不方便（即非ARM的平台就没法用）。作为一个开源的狂热粉丝，我决定以完全自由的方式实现CMSIS-DAP。

STM32F103是一款非常受欢迎，非常便宜的MCU，它有着ARM Cotex-M3的内核，由[STMicroelectronics](https://www.st.com/)设计。正巧我手里有一些叫做[Bluepill](https://satoshinm.github.io/blog/171212_stm32_blue_pill_arm_development_board_first_look_bare_metal_programming.html)的最小系统板，它上面搭载了一片小巧的STM32F103C8T6芯片，并且上面也没有多余的无用的外设组件。这块开发板的唯一不足是连接在USB D+引脚的上拉电阻是10K，但按照标准它应该是1.5K，不过它的USB依然能够正常工作，没有出什么问题。

[image]

给STM32创建一个完整的嵌入式开发环境是非常简单的。在ST的官网上，我们可以下载一些免费软件，其中最重要的是[STM32CubeMX](https://www.st.com/en/development-tools/stm32cubemx.html)。这是一个图形化配置工具，帮助用户选择单片机引脚的功能，并自动生成外设初始化代码。我配置了USB、SPI1和USART1，并选择了USB的Custom HID middleware模式。GPIOB10到GPIO15被配置为JTAG调试需要的引脚。GPIOC13用于驱动单片机上的LED灯。但是我不知道怎么在STM32CubeMX中配置[CoreDebug](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/CEGFHDBB.html)和[DWT](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/CEGJGDCJ.html)，它能在软件里面配置吗？

[image]

ST公司也开发了他们自己的JTAG调试器——[STLink](https://www.st.com/en/development-tools/st-link-v2.html)。当然它并不是必要的，你也可以使用J-Link或者其他种类的调试器。STLink的驱动和程序可以在[ST官网](https://www.st.com/en/development-tools/stsw-link004.html)上下载。在网站里还有一个基于[Eclipse开发环境](https://www.eclipse.org/)开发的[IDE](https://www.st.com/en/development-tools/truestudio.html)，STM32CubeMX也被包含其中。我选择的IDE是基于[CodeBlocks](http://www.codeblocks.org/)的[Embitz](https://www.embitz.org/)，IDE中的[arm_none_eabi_gcc](https://launchpad.net/gcc-arm-embedded)版本是5.4.1。在我完成我的CMSIS-DAP之前，我必须使用STLink来调试我的代码。现在我在使用新做出来的CMSIS-DAP结合[OpenOCD](http://openocd.org/)进行日常的开发。

# 从CMSIS-DAP的源码开始
CMSIS V4 和 V5 中都有CMSIS-DAP的源码，我从[Github](https://github.com/ARM-software/CMSIS_5)上下载了CMSIS_5。CMSIS-DAP的核心文件有五个，但示例其实比核心代码更重要。我将从示例V1的头文件开始分析。

* DAP_config.h

我选择LPC-Link-II V1作为我的参考是因为它是通过USB HID实现的（V2是通过WinUSB实现）。我分析的第一个文件是`DAP_Config.h`。第一个关键位置如下：
```c
#ifdef _RTE_
#include "RTE_Components.h"
#include CMSIS_device_header
#else
#include "device.h"                             // Debug Unit Cortex-M Processor Header File
#endif
```

我想抛弃RTE的相关文件，所以`_RTE_`是不会被定义的，并且我必须创建我自己的`device.h`。

我将参数`CPU_CLOCK`重定义为`72000000`(72MHz)。根据文件里的注释，参数`DAP_PACKET_SIZE`必须重新定义为`64U`。我把`SWO_UART`改为0，这让我的工作轻松不少。参数`TIMESTAMP_CLOCK`也要重定义为`72000000`。LPC-Link-II使用Cortex-M3 的 DWT模块实现时间戳（TIMESTAMP），这也是为什么我想在CubeMX中尝试配置STM32F103的DWT。最后我自己写了一小段代码来实现这个功能（在`device.c`中）：
```c
CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
/**
  * On Cortex-M7 core there is a LAR register in DWT domain.
  * Any time we need to setup DWT registers, we MUST write
  * 0xC5ACCE55 into LAR first. LAR means Lock Access Register.
  */
DWT->CYCCNT = 0;
DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;
```

我定义的引脚和NXP LPCxx的完全不同。我为STM32F103重写了所有的引脚的操作代码。在`DAP_Config.h`这个文件中还有一些奇怪的地方，比如：
```c
// SWCLK/TCK I/O pin -------------------------------------
 
/** SWCLK/TCK I/O pin: Get Input.
\return Current status of the SWCLK/TCK DAP hardware I/O pin.
*/
__STATIC_FORCEINLINE uint32_t PIN_SWCLK_TCK_IN  (void) {
  return ((LPC_GPIO_PORT->PIN[PIN_SWCLK_TCK_PORT]>> PIN_SWCLK_TCK_BIT) & 1U);
}
```

我不明白为什么DAP的代码里需要读取`SWCLK/TCK`引脚的电平，这个引脚明明是被配置为推挽输出来产生时钟信号输送给JTAG从机的。从上面列出来的代码可以看出，它是希望返回当前引脚的电平值，我的实现方式稍微有点不同：
```c
__STATIC_FORCEINLINE uint32_t PIN_SWCLK_TCK_IN  (void) {
  return (uint32_t)(JTAG_TCK_GPIO_Port->ODR & JTAG_TCK_Pin ? 1:0);
}
```

我返回的是当前引脚的输出值。我不确定这是否正确。在整个代码中，这句话只被`DAP.c`中的一个叫`DAP_SWJ_Pins`的函数调用了两次。我猜测`DAP_SWJ_Pins`这个函数是用来测试IO口是否工作正常的。

[... 372 lines cut ...]

> 这里的空循环`while (–count);`会被GCC优化。我在[StackOverflow](https://stackoverflow.com/questions/7083482/how-to-prevent-gcc-from-optimizing-out-a-busy-wait-loop)中找到了一个好点子，它能正常工作但不是太合适，你有更好的方法吗？
```
__STATIC_FORCEINLINE void PIN_DELAY_SLOW (uint32_t delay) {
  uint32_t count;
 
  count = delay;
  while (--count) {
    /**
      * Empty loop will be totally omitted by GCC.
      * Search "How to prevent GCC from optimizing out a busy wait loop?" @ StackOverflow.
      * This solution isn't portable. Zach Lee
      */ 
    __ASM("");
  }
}
```

# 如何使用这个CMSIS-DAP
硬件连接非常简单。我使用一块STM32F103C8T6作为CMSIS-DAP，另一块相同的开发板作为被调试的目标板，下图是两块开发板之间的连线，只需要四根线。

[image]

# 一些有用的链接
* [Another implementation of CMSIS-DAP with SWO support](https://github.com/x893/CMSIS-DAP)

LUCKand, 2019-10-18:
点赞👍

FENYUN323, 2019-11-15:
tql

FENYUN323, 2019-11-15:
tql

FENYUN323, 2019-11-15:
tql

FENYUN323, 2019-11-15:
tql

FENYUN323, 2019-11-15:
tql

FENYUN323, 2019-11-15:
tql

FENYUN323, 2019-11-15:
tql

LIGHT1213, 2020-04-05:
win10有设备插入提示音但是找不到这个设备2333

imuncle, 2020-04-08:
@LIGHT1213 这情况我没遇到过，有可能是Windows识别成了其他类型的设备，可以试试这个文章中的办法：[keil无法识别CMSIS-DAP的问题](https://blog.csdn.net/wangliang0633/article/details/102480525)，希望对你有帮助

LIGHT1213, 2020-04-08:
我仔细看了下，我应该不是这个问题，我这边别识别成了usb输入设备，程序卡在那个USBD_Configured地方了，我再看看把，不过还是谢了

LIGHT1213, 2020-04-09:
@imuncle 我知道怎么回事了，现在我这边keil识别cmsis dap 他的define USBD_PRODUCT_STRING_FS这个东西的字符串里必须带有“CMSIS-DAP”这一部分

imuncle, 2020-04-09:
@LIGHT1213 噢对对对，原文作者没写我也就忘加了，这一点在官方手册里就有写：[Configure USB Peripheral](https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__ConfigUSB__gr.html)

> The String Settings - Product String must contain "CMSIS-DAP" somewhere in the string. This is used by the debuggers to identify a CMSIS-DAP compliant Debug Unit that is connected to a host computer.