imuncle/imuncle.github.io #84: OLED简单GUI开发

imuncle, 2019-10-19:
# OLED简介
我平时最喜欢玩的是中景园的0.96寸的单色OLED，有IIC通信和SPI通信两款，上面的驱动芯片是SSD1306，用起来非常方便。

[image]

该OLED的分辨率是128×64，显示内容时纵坐标上都是以页(page)为单位进行操作，所以这个0.96寸的OLED纵向为8个page。若有一个图标占用了2个page，那么在这个图标2个page的上下空白部分，不能显示其它内容。因为要在这个page的空白部分显示其它内容的话，会擦除这个page上已存在的内容。

有一种方法可以实现在空白部分显示其它内容，就是用并行接口去控制点阵屏，并行接口可以读取屏RAM的显示内容。我们进行写page之前，先把page里面的数据读出来，进行或操作之后再写，这样就保留了之前的图标内容。这种方法会占用MCU很多的IO口。更关键的是，我并没有并行接口，我只有IIC或者SPI，不支持读取屏上某一点的显示内容。

# GUI
图形用户界面（Graphical User Interface，简称 GUI，又称图形用户接口）是指采用图形方式显示的计算机操作用户界面。目前网上已经有很多成熟的专门针对嵌入式系统的GUI，比如emWIN，MicroWindows，MiniGui，ZLG AWTK等。

实现GUI的基础就是实现画点、画线、画圆、画矩形、填充、显示文字等操作，所以下面我就具体讲怎么实现这些基本功能。

# OLED GUI
我使用的办法是在MCU里面创建一个和屏大小相同的数组，不过因为OLED屏上每一个像素点只有两个状态，亮或不亮，所以我完全可以用一位二进制数来表示，所以我使用了128×8的数组来存储OLED的像素点数据。所有的GUI操作均是直接操作这个数组，最后再统一将这个数组传给OLED显示出来，效率也提高了不少。

```c
uint8_t OLED_GRAM[128][8];
```

## 操作点
直接上代码
```c
//画点
//x:0~127
//y:0~63
//t:1 填充 0 清空
void OLED_DrawPoint(uint8_t x,uint8_t y,uint8_t t)
{
    uint8_t pos,bx,temp=0;
    if(x>127||y>63)return;//超出范围了
    pos=y/8;
    bx=y%8;
    temp=1<<bx;
    if(t)OLED_GRAM[x][pos]|=temp;
    else OLED_GRAM[x][pos]&=~temp;
}
```

读取某点的像素值：
```c
uint8_t  OLED_ReadPoint(uint8_t x, uint8_t y)
{
    uint8_t pos,bx,temp=0;
    if(x>127||y>63)return 0;//超出范围了
    pos=y/8;
    bx=y%8;
    temp=1<<bx;
    return OLED_GRAM[x][pos]&temp;
}
```

## 画线
* 画水平线
```c
void  OLED_HLine(uint8_t x0, uint8_t y0, uint8_t x1, uint8_t color)
{
    uint8_t  temp;
    if(x0>x1)
    {
        temp = x1;
        x1 = x0;
        x0 = temp;
    }
    do
    {
        OLED_DrawPoint(x0, y0, color); 
        x0++;
    }
[... 445 lines cut ...]
         if(*str=='\0') break;
         
         GUI_PutChar(bak, bak1, *str++);
         bak += 6;
      }      
   }
   
   return(1);

}
```

调用如下：
```c
uint8_t head[]={"Win_XP"};
uint8_t text[]={"Well Done !"};

Win_Dis(10,10,100,45,head,text);
```

最终效果如下：
[image]

# 参考
* [OLED屏实现任意位置绘制图形](https://blog.csdn.net/lagran/article/details/98893270)

LUCKand, 2019-10-19:
舒服了😍

imuncle, 2019-11-12:
找到一个很强大的OLED GUI库，[u8g2](https://github.com/olikraus/u8g2/)，以及有一篇挺详细的API解读：[玩转u8g2 OLED库，一篇就够](https://blog.csdn.net/dpjcn1990/article/details/92831760)

keven-chain, 2020-01-29:
这个很棒，但是不知道为什么移植到51单片机上就不行了0.0

imuncle, 2020-01-29:
> 这个很棒，但是不知道为什么移植到51单片机上就不行了0.0

@keven-chain 可能是超出了51单片机的内存限制