imuncle/imuncle.github.io #97: 基于STM32 CDC模拟CH340

imuncle, 2019-11-29:
之前写过一篇使用STM32虚拟串口功能的文章：[实现USB CDC通信](https://imuncle.github.io/content.html?id=59)，但是这个有个很大的问题，它的Windows驱动的数字签名过期了，我在我的电脑里搜索了一下，发现有两个驱动：

[image]

不过很可惜，这两个驱动都过期了：

[image]

这就直接导致在Windows上使用ST自己的虚拟串口需要强制跳过数字签名这一步，而每次电脑重启之后Windows就会恢复默认设置，最麻烦的是每次还必须通过重启设置，不过Linux下倒没这个问题，因为数字签名是Windows自己搞出来的东西。

但还是很烦，所以我决定抛弃ST官方的虚拟串口驱动，正好我找到了别人用STM32模拟CH341的代码：[blackmiaool/STM32_USB_CH341 ](https://github.com/blackmiaool/STM32_USB_CH341)，但这已经是五六年前的代码了，当时还是标准库，所以我决定把它用HAL库实现。

踩了一些坑，这玩意儿花了我四天时间，主要还是对USB协议不太熟悉，下面就按照我踩坑的时间顺序记录。

# 第一天：让电脑识别为CH340
这一步很简单，只需要改变设备描述符就行了，具体更改如下：
## 使用STM32CubeMX生成代码
[image]

这里修改以下PID和VID，然后字符串名称就随便写，点击生成代码。

## 修改设备描述符
在`usbd_desc.c`里面，修改`USBD_FS_DeviceDesc`变量：
```c
__ALIGN_BEGIN uint8_t USBD_FS_DeviceDesc[USB_LEN_DEV_DESC] __ALIGN_END =
{
  0x12,   /* bLength */
	0x01,     /* bDescriptorType */
	0x10,
	0x01,   /* bcdUSB = 1.10 */
	0xff,   /* bDeviceClass: CDC */
	0x00,   /* bDeviceSubClass */
	0x00,   /* bDeviceProtocol */
	0x40,   /* bMaxPacketSize0 */
	0x86,
	0x1a,   /* idVendor = 0x1A86 */
	0x23,
	0x75,   /* idProduct = 0x7523 */
	0x63,
	0x02,   /* bcdDevice = 2.00 */
	1,              /* Index of string descriptor describing manufacturer */
	2,              /* Index of string descriptor describing product */
	1,              /* Index of string descriptor describing the device's serial number */
	0x01    /* bNumConfigurations */
};
```
## 修改设备配置描述符
然后修改`usbd_cdc.c`文件里面的`USBD_CDC_CfgFSDesc`变量（因为我配置的Full Speed，如果是其他速度就修改对应的变量就行）：
```c
__ALIGN_BEGIN uint8_t USBD_CDC_CfgFSDesc[0x27] __ALIGN_END =
{
  /*Configuation Descriptor*/
	0x09,   /* bLength: Configuation Descriptor size */
	0x02,      /* bDescriptorType: Configuration */
	0x27,       /* wTotalLength:no of returned bytes */
	0x00,
	0x01,   /* bNumInterfaces: 1 interface */
	0x01,   /* bConfigurationValue: Configuration value */
	0x00,   /* iConfiguration: Index of string descriptor describing the configuration */
	0x80,   /* bmAttributes: self powered */
	0x30,   /* MaxPower 0 mA */
	/*Interface Descriptor*/
	0x09,   /* bLength: Interface Descriptor size */
	0x04,  /* bDescriptorType: Interface */
	/* Interface descriptor type */
	0x00,   /* bInterfaceNumber: Number of Interface */
	0x00,   /* bAlternateSetting: Alternate setting */
	0x03,   /* bNumEndpoints: One endpoints used */
	0xff,   /* bInterfaceClass: Communication Interface Class */
	0x01,   /* bInterfaceSubClass: Abstract Control Model */
[... 273 lines cut ...]
			USBD_CtlSendData(pdev, (uint8_t *)&zero, req->wLength);
			break;
	}
	return;
}
```

* ch340.h
```c
#ifndef CH340_H
#define CH340_H

#include "usbd_def.h"
#include "usbd_cdc.h"

#define CH341_MODEM_OUT 			 0xA4
#define CH341_REQ_READ_REG     0x95
#define CH341_VERSION		 			 0x5F

void CH340_Requset_Handle(USBD_HandleTypeDef *pdev, USBD_CDC_HandleTypeDef *hcdc, USBD_SetupReqTypedef *req);

#endif
```

先吃饭……

zhifangs, 2021-02-26:
可以使用。不错。发现几个bug 但我不会修复。只能正常发送14个字节。多了会导致数据出错，并在重新上电前不可还原正常状态。

leovs, 2022-08-04:
linux 好像无法正常驱动

blackmiaool, 2024-02-21:
好详细